Microchip-MPLAB-Harmony/net #2: If the user used MII connection, an error returns

gpmanly, 2019-07-23:
https://github.com/Microchip-MPLAB-Harmony/net/blob/4bc0fb9d3c913a97004f3b60743b862b656d9bd6/driver/ethphy/src/dynamic/drv_extphy_ksz8041.c#L64

If the user used MII connection (PIC32 is physically MII connected to KSZ8041 and FMIIEN = 1, FETHIO = 1), this line would return error.

Shouldn't it be 
`return (cFlags & DRV_ETHPHY_CFG_MII) ? DRV_ETHPHY_RES_OK : DRV_ETHPHY_RES_CFG_ERR;`

adrian-aur, 2019-07-23:
This code is valid for the KSZ8041RNL which is RMII only.
For the KSZ8041NL the code needs to be adjusted - actually we'll probably add a separate driver for it.
The problem here is that I don't see how the PHY can be configured at run time between the MII/RMII modes. It looks like this can be done using the strap options only. So the version for the 8041NL will have to be hard-coded too. 
For now please use your workaround until we add the 8041NL support.
Thank you for finding and reporting this issue.

gpmanly, 2019-07-24:
@adrian-aur 
That's great to hear. Thanks!

closed 2019-07-24 by gpmanly