Microchip-MPLAB-Harmony/net #4: Unable to add ETHMAC component for PIC32MX795F512H

gGerret, 2019-10-20:
<Configuration Database>[Error]: Failed to activate component: drvPic32mEthmac
<Log>[Error]: Exception cause: Traceback (most recent call last):
  File "C:\microchip\harmony3\net\driver/ethmac/config/drv_intmac_ethmac.py", line 329, in instantiateComponent
    tcpipEthmacIrq_index = int(getIRQnumber("ETHERNET"))
  File "C:\microchip\harmony3\net\driver/ethmac/config/drv_intmac_ethmac.py", line 563, in getIRQnumber
    return irq_index
UnboundLocalError: local variable 'irq_index' referenced before assignment

adrian-aur, 2019-10-21:
There are no demos with PIC32MX as part of the H3 release, as the MX architecture is not yet supported.
We're trying to add support for it in the coming 3.5 release which will be available in 2-3 weeks at the most.

Stui13, 2019-11-13:
Is there any update on this as I have exactly the same issue.

adrian-aur, 2019-11-14:
We're preparing the v3.5 release. It will be public next week and it will have a PIC32MX project example.

Stui13, 2019-11-25:
V3.4 has just been installed and it has the same issue

closed 2019-12-06 by adrian-aur