Microchip-MPLAB-Harmony/net #5: Compile error using TCP IPv4 without DHCP

dchirgwin, 2019-12-04:
Configuring Harmony to use TCP and IPv4 but not DHCP causes a compiler error in the generated code. The code refers to a function in the DHCP module, but does not #include the necessary header file.
See error report below:
`"C:\Program Files (x86)\Microchip\xc32\v2.30\bin\xc32-gcc.exe" -g -x c -c -mprocessor=32MX695F512H -ffunction-sections -O1 -I../src/config/default/library -I../src/config/default/library/tcpip/src -I../src/config/default/library/tcpip/src/common -I../src -I../src/config/default -I../src/packs/PIC32MX695F512H_DFP -I../src/mips -I../src/third_party/rtos/FreeRTOS/Source/portable/MPLAB/PIC32MX -I../src/third_party/rtos/FreeRTOS/Source/include -Werror -Wall -MMD -MF build/default/production/_ext/992804092/drv_memory_file_system.o.d -o build/default/production/_ext/992804092/drv_memory_file_system.o ../src/config/default/driver/memory/src/drv_memory_file_system.c -DXPRJ_default=default -no-legacy-libc 
../src/custom_http_net_app.c: In function 'TCPIP_HTTP_Print_config_dhcpchecked':
../src/custom_http_net_app.c:1608:5: error: implicit declaration of function 'TCPIP_DHCP_IsEnabled' [-Werror=implicit-function-declaration]
     if(TCPIP_DHCP_IsEnabled(hNet))
     ^
cc1.exe: all warnings being treated as errors
`

dchirgwin, 2019-12-04:
PR raised with a fix for this issue.

NiyasMCHP, 2019-12-05:
Thank you for reporting the issue. 
Currently, DHCP Client is required in project graph for the proper functioning of application file custom_http_net_app.c. So please add DHCP Client to project graph.

To fix this issue, we will enforce the addition of DHCP client with IPv4 in the next release.

dchirgwin, 2019-12-05:
Thanks for your suggestion. Unfortunately, DHCP is not suitable for my application. The situation is that we have an embedded controller which is monitored and managed using a web browser. The embedded controller needs to have a static IP address otherwise the operator doesn't know what URL to use to talk to it.
Maybe I have misunderstood you - are you saying that DHCP needs to be included even if it isn't used? I guess that would work, but it's a shame to include not only DHCP but also UDP, which are not needed and not used.
As far as I can tell, the only dependency on DHCP is the function `TCPIP_DHCP_IsEnabled()`. 
I'm just a newbie with the Harmony library, and may be missing something important, so please can you help me to understand why my proposed fix is not suitable?

NiyasMCHP, 2019-12-05:
I understood your requirement. We will fix this issue of build error when DHCP is not enabled, in the next release. 

Please perform following steps as a temporary workaround,
- Copy dhcp.c from net/tcpip/src to the project folder and copy dhcp.h from net/tcpip to the  project folder. 
- Add these file to the MPLAB project. 
- Also add following line of code in tcpip.h file
#include "tcpip/dhcp.h"

This will enable you to work with your HTTP demo without enabling the DHCP in your project.

NiyasMCHP, 2019-12-05:
I forgot to answer your following query.
"I'm just a newbie with the Harmony library, and may be missing something important, so please can you help me to understand why my proposed fix is not suitable?"

Adding just #include "tcpip/dhcp.h" will not resolve the issue, as the implementation of  TCPIP_DHCP_Enable() function is in dhcp.c file.
You need to perform all the 3 steps mentioned in previous message to fix the issue.

dchirgwin, 2019-12-06:
Thank you for that explanation - it really helps me to understand what's going on! 

I was surprised to see that my project compiled OK without `dhcp.c` being added, so I looked a bit deeper. It appears that the DHCP module is not actually used because the linker optimises out the function `TCPIP_HTTP_Print_config_dhcpchecked()` because there are no callers. However, the function *is* in fact called, but the linker doesn't realize because it a dynamicPrint function and is only called through a function pointer, not directly.
So in the end I can now see why the DHCP module is necessary. Thanks again for your help!

dchirgwin, 2019-12-06:
P.S. Just for information for anyone who has the same issue, I have now found that the dependency on DHCP only exists if the HTTP_NET module has Dynamic Variable Processing enabled. For me this is a much better solution, as it avoids the overhead of all the DHCP and UDP code (c. 20K of Flash space), and I don't need the dynamic variables anyway.

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur