Microchip-MPLAB-Harmony/net #8: Compiler error if Dynamic Variable Processing is disabled with no email or DDNS

dchirgwin, 2019-12-06:
## Problem
If the HTTPNET_SERVER module has Dynamic Variable Processing disabled, and the configuration does not include DDNS or Email, then there is a compiler error in the generated code:

```nbproject/Makefile-default.mk:574: recipe for target 'build/default/production/_ext/1360937237/custom_http_net_app.o' failed
../src/custom_http_net_app.c:152:13: error: 'lastSuccess' defined but not used [-Werror=unused-variable]
 static bool lastSuccess = false;
             ^
cc1.exe: all warnings being treated as errors
make[2]: *** [build/default/production/_ext/1360937237/custom_http_net_app.o] Error 1
```

## Diagnosis
`lastSuccess` is **declared** and **initialized** unconditionally

`lastSuccess` is **written** if:
* `#if (HTTP_APP_USE_EMAIL != 0)`
* `#if defined(TCPIP_STACK_USE_DYNAMICDNS_CLIENT)`

`lastSuccess` is **read** if:
* `<#if ((TCPIP_HTTP_NET_DYNVAR_PROCESS?has_content) && (TCPIP_HTTP_NET_DYNVAR_PROCESS  == true))>`

The problem occurs when your configuration is such that `lastSuccess` is not written or read, which is the case if you're not using email, DynamicDNS or Dynamic variable processing.

## Workaround
Simply comment out the declaration (around line 152) in the generated code.

## Potential Solution
One way would be to make the declaration conditional in the same way that it's usage is conditional. To do this, the template would need to be modified something like this:
```
<#if ((TCPIP_HTTP_NET_DYNVAR_PROCESS?has_content) && (TCPIP_HTTP_NET_DYNVAR_PROCESS  == true))>
static bool lastSuccess = false;
<#else>
#if (HTTP_APP_USE_EMAIL != 0) || defined(TCPIP_STACK_USE_DYNAMICDNS_CLIENT)
static bool lastSuccess = false;
#endif
</#if>
```

adrian-aur, 2020-02-28:
Thank you for the report.
The fix has been integrated in the main repository and will be part of the next release.

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur