Microchip-MPLAB-Harmony/net #9: FreeRTOS OSAL configuration conflicting with TCPIP implementation

KonssnoK, 2019-12-09:
Hello,
using 

> #define OSAL_USE_RTOS 1

seems to be incompatible your current tcp implementation.

In tcp.c:
`   OSAL_CRITSECT_DATA_TYPE status = OSAL_CRIT_Enter(OSAL_CRIT_TYPE_LOW);  // Full System Lock

.............

    TCBStubs = (TCB_STUB**)TCPIP_HEAP_Calloc(tcpHeapH, nSockets, sizeof(*TCBStubs));

...........

    OSAL_CRIT_Leave(OSAL_CRIT_TYPE_LOW, status);  // Full System Lock`

Function 
> OSAL_CRIT_Enter
is implemented with double priority argument:
`OSAL_CRITSECT_DATA_TYPE __attribute__((nomips16,nomicromips)) OSAL_CRIT_Enter(OSAL_CRIT_TYPE severity)
{
    switch (severity)
    {
        case OSAL_CRIT_TYPE_LOW:
            /* LOW priority critical sections just disable the scheduler */
            vTaskSuspendAll();
            break;

        case OSAL_CRIT_TYPE_HIGH:
            /* HIGH priority critical sections disable interrupts */
            portENTER_CRITICAL();
            break;
    }
    
    return(0);
}`

In the case above the system is entering a critical section by using vTaskSuspendAll().
After this step the function is calling a HEAP_Calloc initializer, which uses a semaphore as follows:
`OSAL_SEM_Pend(&hDcpt->_heapSemaphore, OSAL_WAIT_FOREVER);`

which gets translated to
`if (xSemaphoreTake(*(SemaphoreHandle_t*)semID, timeout) == pdTRUE)`

which is a
`xQueueGenericReceive( ( QueueHandle_t ) ( xSemaphore ), NULL, ( xBlockTime ), pdFALSE )`

Sadly, in the vTaskSuspendAll documentation the following is written:
> API functions that have the potential to cause a context switch (for example, vTaskDelayUntil(), **xQueueSend**(), etc.) must not be called while the scheduler is suspended.

This is triggering the following assert:
`	#if ( ( INCLUDE_xTaskGetSchedulerState == 1 ) || ( configUSE_TIMERS == 1 ) )
	{
		configASSERT( !( ( xTaskGetSchedulerState() == taskSCHEDULER_SUSPENDED ) && ( xTicksToWait != 0 ) ) );
	}
	#endif`
making the initialization crash.

This can be fixed either by checking each critical section in Harmony not to use OSAL_CRIT_TYPE_LOW or by removing the Low priority subtype in FreeRTOS Osal.

This is the workaround I had to use in my company to make things work:
`OSAL_CRITSECT_DATA_TYPE __attribute__((nomips16, nomicromips)) OSAL_CRIT_Enter(OSAL_CRIT_TYPE severity)
{
	switch (severity) {
		//case OSAL_CRIT_TYPE_LOW:
		//    /* LOW priority critical sections just disable the scheduler */
		//    vTaskSuspendAll();
		//    break;
	case OSAL_CRIT_TYPE_LOW:
	case OSAL_CRIT_TYPE_HIGH:
		/* HIGH priority critical sections disable interrupts */
		portENTER_CRITICAL();
		break;
	}

	return(0);
}`

Cheers

adrian-aur, 2019-12-10:
The sequence that you describe is indeed more than problematic: a semaphore (called by the Calloc function) should never be acquired when the scheduling is disabled. Obviously it makes no sense.

But I do not see what you're describing in the tcp.c.
On line #593, there is:
OSAL_CRITSECT_DATA_TYPE status = OSAL_CRIT_Enter(OSAL_CRIT_TYPE_LOW);  // System Lock
then on line #623 there is:
OSAL_CRIT_Leave(OSAL_CRIT_TYPE_LOW, status);  // System unlock
And only after that, on line #651 I see the allocation:
    TCBStubs = (TCB_STUB**)TCPIP_HEAP_Calloc(tcpHeapH, nSockets, sizeof(*TCBStubs));

So this allocation happens with the scheduling unlocked, as it should, no problem.
I'm looking in the tcp.c for this latest release, v3.5.0.
Please clarify.

KonssnoK, 2019-12-11:
Yes the example i used is taken from an older version indeed.
So old that is not even in the github history...

I guess i have to tell somebody to update...

Thanks!

closed 2019-12-11 by KonssnoK