Microchip-MPLAB-Harmony/net #11: Add Support for KSZ8041 PHY

st-gt, 2020-01-06:
Support for the KSZ8041 PHY appears to have be removed as part of [ab002c0](https://github.com/Microchip-MPLAB-Harmony/net/commit/ab002c017322d56af776332864a5099501cfc2fc#diff-cf88e772de8bec2a515c35e1ae4d0a72) although this support may have never been fully implemeted. Is there a particular reason this was removed/not implemented for MH3?

Many - but not all - of the support files for KSZ8041 (eg. [drv_extphy_ksz8041.c](https://github.com/Microchip-MPLAB-Harmony/net/blob/master/driver/ethphy/src/dynamic/drv_extphy_ksz8041.c) and [drv_extphy_ksz8041.py](https://github.com/Microchip-MPLAB-Harmony/net/blob/master/driver/ethphy/config/drv_extphy_ksz8041.py) appear to still be in place from the initial commits to the repository. What is needed to add support for this into MH3?

NiyasMCHP, 2020-01-07:
The driver for KSZ8041 is very similar to the one for KSZ8061.
The KSZ8041 component is removed from H3 MHC as this PHY is not present on any of the currently supported development board; hence, not tested.

Adding the support for this PHY to H3 is in our plan; but if you have an urgent need to add the KSZ8041 driver, the actions required to add KSZ8041 in MHC  are
1) add KSZ8041 component in net\config\module.py (refer addition of KSZ8061)
2) add script files (.py and .ftl) KSZ8041 in net\driver\ethphy\config (refer script files of KSZ8061)
3) add driver source files at D:\11_Repo\01_H3\h3\net\driver\ethphy\src\dynamic. (this is already available, but need to test)

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur