Microchip-MPLAB-Harmony/net #12: LAN8742 driver

redphoenix741, 2020-01-14:
Hi,

I noticed that LAN8742 is very similar to LAN8740 which has a driver on Harmony.

Is the driver LAN8740 directly compatible with the LNA8742? 

Thank you.

adrian-aur, 2020-01-14:
The main PHY driver is based on the IEEE 802.3 standard and all PHYs implement it.
The differences are in the vendor specific functions that a specific PHY supports.
For a Harmony PHY driver these are:
- PHY driver function to configure the operation mode: MII/RMII (called miiConfigure)
- PHY driver function to configure the MDIX mode (called mdixConfigure)
- PHY driver function to get the SMI clock rate (smiClockGet)
So you can do a quick check to see if the 2 PHYs have compatible registers for MII/RMII and MDIX configuration and support the same MIIM clock.
Then you'll be sure that they are compatible.
Or you can just try the driver and see if it works, no harm can come from that as long as you use only standard functionality.

redphoenix741, 2020-01-15:
Hi,

Thank you very much for the reply.

closed 2020-01-15 by redphoenix741