Microchip-MPLAB-Harmony/net #13: Compile error for DRV KSZ8091

youpko, 2020-01-20:
Net v3.5.1 contains a small error for the KSZ8091 driver code

```
../src/config/default/driver/ethphy/src/dynamic/drv_extphy_ksz8091.c: In function 'DRV_EXTPHY_MDIXConfigure':
../src/config/default/driver/ethphy/src/dynamic/drv_extphy_ksz8091.c:103:5: error: missing braces around initializer [-Werror=missing-braces]
     } vendorData = { 0 };
     ^
../src/config/default/driver/ethphy/src/dynamic/drv_extphy_ksz8091.c:103:5: error: (near initialization for 'vendorData.<anonymous>') [-Werror=missing-braces]
```

adrian-aur, 2020-01-21:
I'm compiling this file with xc32-2.30 and I do not get any warning/error.
Can you please share the compiler version and the settings that you use?

youpko, 2020-01-21:
Sorry forgot to put that information in the original message.
My build enviorment is:
*  IDE: v5.30
*  XC32: v2.30
   * Additional warning
   * Make warnings into errors.
*  NET: v3.5.1

adrian-aur, 2020-01-22:
That's exactly what I use, and I see no error.
Not sure I know what's going on.
If you right click on that file in the project and compile it, do you get the same error?
Please also try a different project/configuration, see if the problem persists.

youpko, 2020-01-22:
If I use right-click compile the message is the same. 
`../src/config/default/driver/ethphy/src/dynamic/drv_extphy_ksz8091.c:103:5: error: missing braces around initializer [-Werror=missing-braces]`

I tried the following:
```
} vendorData = { 0 }; = Error
} vendorData = {{ 0 }}; = Ok
} vendorData = { }; = OK
} vendorData = {.w = 0 }; = OK
```

Changing the following also works (by setting the uint32_t as first union item):
```
union
    {
        uint32_t    w;
        struct
        {
            uint16_t low;
            uint16_t high;
        };
    } vendorData = { 0 };
```

I don't get why my case is special because when I started the project harmony generated everything and the problem has been here from the beginning. Before I even touched any of the settings. 
I just discarded the problem as a typo (which i make plenty of myself) but for good measure I decided to make a post about it anyway.

adrian-aur, 2020-01-23:
Thank you for reporting the issue, it definitely needs to be corrected.
My preference will be to use the version where the uint32_t member is on 1st position.
I'll correct it in the repo and it will be part of the next release.
Still I don't understand why I d not get the error but I'll correct it nevertheless.
Meanwhile please use your fix for this.

closed 2020-01-23 by youpko