Microchip-MPLAB-Harmony/net #16: PIC32MX MAC address gets scrambled

dchirgwin, 2020-02-14:
There is a bug in "driver\ethmac\src\dynamic\drv_eth_pic32_lib.c". The function "DRV_ETH_MACSetAddress()" sets the bytes in the wrong order by transposing EMACxSA0 and EMACxSA2.

The order should be:
EMACxSA0 => bytes 4 & 5
EMACxSA1 => bytes 2 & 3
EMACxSA2 => bytes 0 & 1
This can be seen from the function "DRV_ETH_MACGetAddress()", which uses the correct order.

#Symptoms
In the NETCONFIG instance, set the MAC Address to "10:20:30:40:50:60"
Using Wireshark, the actual MAC transmitted is shown as "50:60:30:40:10:20"

dchirgwin, 2020-02-14:
I have fixed this bug and raised a pull-request.

adrian-aur, 2020-02-26:
Thank you for reporting this. Indeed a bug in that library function. It will be fixed as you suggested.

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur