Microchip-MPLAB-Harmony/net #18: FTP Authorization

youpko, 2020-03-06:
An update or two ago the Telnet server changed it authorization method to a callback method instead of static defined strings for user name and password. I would like to request the same approach for the FTP server module.

The core module also updated recently and it changed the filesystem dependency type in the harmony configuration so some links are missing in my configuration overview.

adrian-aur, 2020-03-06:
Thank you for the good suggestion, we'll add it to the FTP server in a future release.
There will be a new release coming for net fairly soon, and it will use the latest versions from the core repo.  I hope your problems will be solved then.

adrian-aur, 2020-04-23:
The FTP updated authentication has been updated and will be part of the 3.6.0 release.

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur