Microchip-MPLAB-Harmony/net #19: NET_PRES_SocketSignalHandlerDeregister handler issue

lucapascarella, 2020-04-07:
_NET_PRES_SocketSignalHandlerDeregister_ calls the sublayer deregister function passing a wrong parameter. See the below code for a solution.

`bool NET_PRES_SocketSignalHandlerDeregister(NET_PRES_SKT_HANDLE_T handle, NET_PRES_SIGNAL_HANDLE hSig)
{
    NET_PRES_SocketData * pSkt;
    if ((pSkt = _NET_PRES_SocketValidate(handle)) == NULL)
    {
        return false;
    }

    NET_PRES_TransSignalHandlerDeregister fp = pSkt->transObject->fpHandlerDeregister;
    if (fp == NULL)
    {
        pSkt->lastError = NET_PRES_SKT_OP_NOT_SUPPORTED;
        return false;
    }
    // return (*fp)(handler, (NET_PRES_SIGNAL_FUNCTION) hSig); // Wrong call
    return (*fp)(pSkt->transHandle, (NET_PRES_SIGNAL_FUNCTION) hSig); // Correct call

}`

adrian-aur, 2020-04-07:
You are correct. Thank you for reporting this.
The issue is fixed and it will be part of the next release.

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur