Microchip-MPLAB-Harmony/net #20: Incorrect line-end sequence

dchirgwin, 2020-04-09:
In `tcpip_manager.c` there are several places where line ends for `SYS_ERROR_PRINT` and `SYS_CONSOLE_MESSAGE` strings are "\n\r" rather than the standard Windows "\r\n" (which is used everywhere else).

I know this is a really minor point, and I hesitate to raise it, but:
1. It is definitely incorrect
1. It does have annoying consequences (my terminal emulator is showing text incorrectly formatted)
1. It can be fixed in 2 minutes with a search/replace

If you're still reading, you might also want to know that the same occurs in every instance of `exceptions.c` in the whole Harmony project (all 494 of them, not just the `net` repo but all the other repos as well).
The same points apply - it would be a 2-minute fix with a global search/replace, and it seems a shame to leave a known error in place, albeit a very minor one.

adrian-aur, 2020-04-09:
Thank you for reporting this issue.
It may be minor but I agree it's annoying.
I'll take care of it ASAP in the net repo.
Can you please replicate this issue in the other repos where you found this.

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur