Microchip-MPLAB-Harmony/net #21: Missing #include in TCP.H

dchirgwin, 2020-04-15:
# The Problem
The generated file `tcp.h` depends on `tcpip.h` for things like `IP_ADDRESS_TYPE` but it does not include `tcpip.h`.

Therefore, this does not compile:
```
#include "tcpip/tcp.h"

void func()
{
    // Do something with TCP
}
```

The user code has to manually include the pre-requisite, before using `tcp.h`:
```
#include "tcpip.h"
#include "tcpip/tcp.h"

void func()
{
    // Do something with TCP
}
```
# The Solution
Simply add `#include "tcpip.h"` to the template for `tcp.h`.
😄

adrian-aur, 2020-04-15:
To use the TCP/IP stack, the user code needs to have:
#include "tcpip.h"
That's all. Not include the individual protocols. That's how it's supposed to work.

dchirgwin, 2020-04-15:
Ah. Thanks for the explanation.
I can see now that the other protocols are #included at the end of `tcpip.h`.

closed 2020-04-15 by dchirgwin