Microchip-MPLAB-Harmony/net #23: HTTP_NET cache issue with SQI

lucapascarella, 2020-04-23:
Using http_net.c with an SQI SST flash connected to a PIC32 MZ I found a potential issue when webpages are loaded through MPFSUPLOAD.

In `TCPIP_HTTP_NET_IO_RESULT TCPIP_HTTP_NET_FSUpload(TCPIP_HTTP_NET_CONN* pHttpCon)` the malloc call may return a cachable address that is not compatible with SQI writer that corrupts the uploaded image.

`pHttpCon->uploadBufferStart = 0;
            mpfsAllocSize = ((MPFS_UPLOAD_WRITE_BUFFER_SIZE + (SYS_FS_MEDIA_SECTOR_SIZE - 1)) / SYS_FS_MEDIA_SECTOR_SIZE) * SYS_FS_MEDIA_SECTOR_SIZE;
            if((pHttpCon->uploadBufferStart = (uint8_t*)(*http_malloc_fnc)(mpfsAllocSize)) != 0)
            {
// TEMP WORKAROUND
                if (IS_KVA0(pHttpCon->uploadBufferStart))
                    pHttpCon->uploadBufferStart = KVA0_TO_KVA1(pHttpCon->uploadBufferStart);
// TEMP WORKAROUND
                pHttpCon->uploadBufferEnd = pHttpCon->uploadBufferStart + mpfsAllocSize;
                memcpy(pHttpCon->uploadBufferStart, MPFS_SIGNATURE, sizeof(MPFS_SIGNATURE) - 1);
                pHttpCon->uploadBufferCurr = pHttpCon->uploadBufferStart + sizeof(MPFS_SIGNATURE) - 1;
                // Print unmount error, but continue with upload process because the memory may be erased or corrupted
                if (SYS_FS_Unmount(MPFS_UPLOAD_MOUNT_PATH) != SYS_FS_RES_SUCCESS) {
                    char tmp[8];
                    sprintf(tmp, "fs-%d", SYS_FS_Error());
                    _HTTP_Report_ConnectionEvent(pHttpCon, TCPIP_HTTP_NET_EVENT_FS_MOUNT_ERROR, tmp);
                }
                pHttpCon->httpStatus = TCPIP_HTTP_NET_STAT_UPLOAD_WRITE;
                return TCPIP_HTTP_NET_IO_RES_WAITING;
            }`

adrian-aur, 2020-04-23:
I assume that the problem is with the function that actually does the write:
SYS_FS_MEDIA_MANAGER_SectorWrite(   uint16_t diskNum, uint32_t sector, uint8_t ***dataBuffer**,     uint32_t numSectors);
Looking into the description of that API there's no mention that the data buffer needs to be cached/uncached, aligned, etc. Actually the same code works for all SYS_FS media.
Maybe this is a problem with the SQI driver, I'll report it  and see what the reply is.

lucapascarella, 2020-04-23:
Yes indeed. The SQI peripheral of the PIC32MZ works in three different ways. The one used by SYS_FS_MEDIA_MANAGER_SectorWrite uses the internal DMA module that cannot read cached values.

adrian-aur, 2020-07-02:
Solved in net 3.6.0

closed 2020-07-02 by adrian-aur