Microchip-MPLAB-Harmony/net #24: LAN9303 support seems incomplete

harkwoodservices, 2020-05-19:
I can see various source files for the LAN9303 within the net ethphy files, but it doesn't seem to be complete.  It doesn't show up in MHC and looking at the various config files details for it are in some but not others.

Is the LAN9303 available as it was in Harmony v2 ?

I'm looking to refresh a project written using Harmony2, everything seems to be there apart from the LAN9303 support.

adrian-aur, 2020-05-19:
MHC support for LAN9303 will be added in the coming net release, available in few weeks from now.

harkwoodservices, 2020-05-19:
Excellent, thanks.  That should give me time to get to grips with it before attacking the migration in earnest.

closed 2020-05-19 by harkwoodservices