Microchip-MPLAB-Harmony/net #25: Bug in HTTP_NET error reporting

dchirgwin, 2020-06-11:
I have encountered something that looks very much like a bug.
```
static uint32_t _HTTP_ConnectionStringFind(TCPIP_HTTP_NET_CONN* pHttpCon, const char* str, uint16_t startOffs, uint16_t searchLen)
{
    uint16_t    peekOffs, peekReqLen, peekSize, avlblBytes;
    char*   queryStr;

    char* srchBuff = (char*)(*http_malloc_fnc)(httpPeekBufferSize + 1);
    if(srchBuff == 0)
    {
        _HTTP_Report_ConnectionEvent(pHttpCon, TCPIP_HTTP_NET_EVENT_PEEK_ALLOC_BUFFER_ERROR, (void*)(uint32_t)httpPeekBufferSize);
        return 0xffff;
    }
```

This code attempts to allocate a buffer, and if it fails, it tries to report the error. The problem is the void pointer to the buffer size that is returned as the 3rd parameter of _HTTP_Report_ConnectionEvent()

This eventually calls TCPIP_HTTP_NET_EventReport() which casts the void pointer to a string and tries to print it. 

```
void TCPIP_HTTP_NET_EventReport(TCPIP_HTTP_NET_CONN_HANDLE connHandle, TCPIP_HTTP_NET_EVENT_TYPE evType, const void *evInfo, const struct _tag_TCPIP_HTTP_NET_USER_CALLBACK *pCBack)
{
    char *evMsg = (char *)evInfo;

    if(evType < 0)
    {   // display errors only
        if(evMsg == 0)
        {
            evMsg = "none";
        }
        SYS_CONSOLE_PRINT("HTTP event: %d, info: %s\r\n", evType, evMsg);
    }
}

```
In my configuration, the (default) buffer size is 512 bytes, so the code attempts to print a string starting at address 0x00000200 and gets a CPU exception type 7 (Data Bus Error).

Obviously this only happens if your `malloc()` fails, so it won't affect most projects, but it seems evidently incorrect.

Meanwhile, I need to work out why I'm running out of heap space in the first place. 😄

adrian-aur, 2020-06-12:
Yes, you're right, the simple application function offered as an example: TCPIP_HTTP_NET_EventReport() doesn't take into account the event type but blindly assumes that the evInfo points to a string.
While the majority of events carry printable/string info, not all of them do. And the allocation size that failed is an integer in this case.
There's different ways to fix this and I'll have to see if we can integrate the fix in this coming/imminent release.
Thank you for reporting this issue.

adrian-aur, 2020-07-02:
Fixed in net v3.6.0

closed 2020-07-02 by adrian-aur