Microchip-MPLAB-Harmony/net #27: Adding TCP/IP Application Layer configurator makes it impossible to save the configuration

zip639, 2021-01-12:
When I add the "TCP/IP Application Layer configurator" into the project graph I cannot save the project anymore.  The error message is : "Not able to save MHC state. Please ensure that MHC state files are not open in another application".

I pretty sure that no other application uses the MHC files.

After close the Harmony 3 configurator window I cannot restart it: "MPLAB Harmony Configurator could not find required files in this project configuration."

Here are the info: 
 - MPLAB X IDE v5.45
 - Harmony 3 Configurator v3.6.2
 - µC: SAME54P20A
 - OS: Xubuntu 18.04

The mhc.log file contains this trace: 

```
<Log>[Error]: Exception cause: java.io.UncheckedIOException: java.nio.file.NoSuchFileException: /media/sf_Data/MPLab_workspace/ethernet/udp_server/server/firmware/src/config/default/default.mhc.bak/TCP_IP_STACK
<Log>[Error]: Stack trace:
<Log>[Error]: java.nio.file.FileTreeIterator.fetchNextIfNeeded(FileTreeIterator.java:88)
<Log>[Error]: java.nio.file.FileTreeIterator.hasNext(FileTreeIterator.java:104)
<Log>[Error]: java.util.Iterator.forEachRemaining(Iterator.java:115)
<Log>[Error]: java.util.Spliterators$IteratorSpliterator.forEachRemaining(Spliterators.java:1801)
<Log>[Error]: java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:482)
<Log>[Error]: java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:472)
<Log>[Error]: java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
<Log>[Error]: java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
<Log>[Error]: java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
<Log>[Error]: java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:485)
<Log>[Error]: com.microchip.h3.database.SaveCommand.copyFolder(SaveCommand.java:190)
<Log>[Error]: com.microchip.h3.database.SaveCommand.restoreBackupState(SaveCommand.java:133)
<Log>[Error]: com.microchip.h3.database.SaveCommand.saveState(SaveCommand.java:50)
<Log>[Error]: com.microchip.h3.database.ConfigurationDatabase.saveState(ConfigurationDatabase.java:455)
<Log>[Error]: com.microchip.mh3.mplx_launcher.core.CoreIDE.save(CoreIDE.java:919)
<Log>[Error]: com.microchip.mh3.window.standalone.MainWindow.fileSave(MainWindow.java:1179)
<Log>[Error]: com.microchip.mh3.window.standalone.MainWindow.fileSaveItemActionPerformed(MainWindow.java:700)
<Log>[Error]: com.microchip.mh3.window.standalone.MainWindow.access$1100(MainWindow.java:68)
<Log>[Error]: com.microchip.mh3.window.standalone.MainWindow$11.actionPerformed(MainWindow.java:512)
<Log>[Error]: javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
<Log>[Error]: javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
<Log>[Error]: javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
<Log>[Error]: javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
<Log>[Error]: javax.swing.AbstractButton.doClick(AbstractButton.java:376)
<Log>[Error]: javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
<Log>[Error]: javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
<Log>[Error]: java.awt.Component.processMouseEvent(Component.java:6539)
<Log>[Error]: javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
<Log>[Error]: java.awt.Component.processEvent(Component.java:6304)
<Log>[Error]: java.awt.Container.processEvent(Container.java:2239)
<Log>[Error]: java.awt.Component.dispatchEventImpl(Component.java:4889)
<Log>[Error]: java.awt.Container.dispatchEventImpl(Container.java:2297)
<Log>[Error]: java.awt.Component.dispatchEvent(Component.java:4711)
<Log>[Error]: java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
<Log>[Error]: java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
<Log>[Error]: java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
<Log>[Error]: java.awt.Container.dispatchEventImpl(Container.java:2283)
<Log>[Error]: java.awt.Window.dispatchEventImpl(Window.java:2746)
<Log>[Error]: java.awt.Component.dispatchEvent(Component.java:4711)
<Log>[Error]: java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
<Log>[Error]: java.awt.EventQueue.access$500(EventQueue.java:97)
<Log>[Error]: java.awt.EventQueue$3.run(EventQueue.java:709)
<Log>[Error]: java.awt.EventQueue$3.run(EventQueue.java:703)
<Log>[Error]: java.security.AccessController.doPrivileged(Native Method)
<Log>[Error]: java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
<Log>[Error]: java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
<Log>[Error]: java.awt.EventQueue$4.run(EventQueue.java:733)
<Log>[Error]: java.awt.EventQueue$4.run(EventQueue.java:731)
<Log>[Error]: java.security.AccessController.doPrivileged(Native Method)
<Log>[Error]: java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
<Log>[Error]: java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
<Log>[Error]: org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
<Log>[Error]: java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
[... 42 lines cut ...]
<Log>[Error]: java.awt.Container.dispatchEventImpl(Container.java:2283)
<Log>[Error]: java.awt.Window.dispatchEventImpl(Window.java:2746)
<Log>[Error]: java.awt.Component.dispatchEvent(Component.java:4711)
<Log>[Error]: java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
<Log>[Error]: java.awt.EventQueue.access$500(EventQueue.java:97)
<Log>[Error]: java.awt.EventQueue$3.run(EventQueue.java:709)
<Log>[Error]: java.awt.EventQueue$3.run(EventQueue.java:703)
<Log>[Error]: java.security.AccessController.doPrivileged(Native Method)
<Log>[Error]: java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
<Log>[Error]: java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
<Log>[Error]: java.awt.EventQueue$4.run(EventQueue.java:733)
<Log>[Error]: java.awt.EventQueue$4.run(EventQueue.java:731)
<Log>[Error]: java.security.AccessController.doPrivileged(Native Method)
<Log>[Error]: java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
<Log>[Error]: java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
<Log>[Error]: org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:159)
<Log>[Error]: java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
<Log>[Error]: java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
<Log>[Error]: java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
<Log>[Error]: java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
<Log>[Error]: java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
<Log>[Error]: java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

Any help would be appreciated.

NiyasMCHP, 2021-01-13:
We couldn't reproduce this issue. Could you please restart MPLAB and MHC to see whether the issue persists?
If it persists, please provide more detailed steps to reproduce the issue at our end.

zip639, 2021-01-13:
Restarting MPLAB and MHC does not have any effect. 

Unfortunatly don't think I have done any special step. 
Here is are the steps :
- Open MPLAB
- Create a new MPLAB Harmony 3 project
- Indicate the path of the MHC framework
- Set ATSAME54P20A as target device
- The project is created and the MHC launches.
- Add TCP/IP application layer configurator
- Try to save the MHC configuration

I just tried to downgrade with net v.3.6.3 and I have the same error.

NiyasMCHP, 2021-02-17:
Sorry for the delay; We couldn't reproduce the issue on our test setup.
I have asked our internal MHC team to further investigate this issue.
In the meantime, do let us know if you have any other observations.

sandeepm1, 2021-02-17:
"Not able to save MHC state. Please ensure that MHC state files are not open in another application".

This error message comes when ".mhc" folder is open in File explorer or any component yml is open in another application. When we save mhc config it removes the old configuration files and saves new one. 
Before saving the config please close other applications even File explorer.

zip639, 2021-02-17:
Okay, I managed to get more information about the issue. MPLAB is in fact running in a VirtualBox VM.
The VM's guest OS is Xubuntu on which MPLab is installed. There is a shared folder between the guest OS and the host OS (Window 10). 
It appears that when saving the project on that shared folder, the MHC cannot save but only when using using the TCP/IP stack. However, it is perfectly working when not saving in that shared folder.

Another information : when the issue shows up, in `firmware/src/config/project_name/` is created a a folder named `project_name.mhc.bak`which seems to contains the MHC configuration.

adrian-aur, 2021-05-18:
MHC team will provide a fix.

closed 2021-05-18 by adrian-aur

chadmawhinney, 2024-07-23:
Where is the fix?  I'm having this issue in 2024.