Microchip-MPLAB-Harmony/net #28: ICMP Echo Responses Ignore Routing Table

DavidIngraham, 2021-03-09:
In the ICMP driver, echo responses always reply directly to the SRC MAC of the echo request packet. ICMP is a layer 3 protocol and should use appropriate IP routing rules when sending responses.

This is particularly problematic on subnets which use asymmetric routes for ingress and egress. If incoming packets come in from one router, and outgoing packets must egress through a different router, echo responses will always end up at the ingress router.

adrian-aur, 2021-03-10:
Yes, just resending the packet where it came from could create issues in certain network configurations.
We'll create an internal case for this and the fix will be part of the next release.
Thank you for reporting this.

adrian-aur, 2021-05-18:
The fix is part of net v3.7.2 release

closed 2021-05-18 by adrian-aur