Microchip-MPLAB-Harmony/net #30: MALLOC Question

Skaptor, 2021-06-16:
Hello this is not an issue, rather a question regarding the malloc usage in the TCP/IP stack. I'm currently using heap4 in freeRTOS in a PIC32MZ. Why is it using the standard malloc instead of the "safer" OS alternative?

adrian-aur, 2022-02-21:
The TCP/IP stack uses memory allocation just at the initialization to allocate its own heap. After that it will use only its  own private heap. This will work fine as long as you initialize the stack at system start up and then just use it.
If you intend to turn it down and then up again at run time, then definitely a thread safe allocation function should be used.
And that's why you can specify the allocation function when you initialize the stack.

adrian-aur, 2022-04-11:
I see the article you mention. It is written by users of the TCP/IP stack on Wi-Fi platforms.
The overhead of an internal heap isn't too big, the allocation routines are relatively small. I think it should be comparable with the external one. But I haven't really done a comparison for this.
Even for the external heap there is some internally maintained data.

What's said above is true for the internal heap, yes. Only then at the stack initialization it tries to allocate its own heap. 
When using an external heap the stack will use it directly and it will create a semaphore for multi-threaded protection, if the flag TCPIP_STACK_HEAP_FLAG_NO_MTHREAD_SYNC is cleared.

closed 2023-08-21 by adrian-aur