Microchip-MPLAB-Harmony/net #31: Doesn't work with newer versions, and steps and pictures out of sequence.

mhoover-fdi, 2022-01-04:
I used Microchips Video on a PIC32 Ethernet Starter Kit II using Harmony v2.06 and done just fine. I tried to use the video along with this link that microchip support gave. We have used MLA libraries for a while and thinking going the Harmony 3 and PIC32 direction for some things.

This migration might be valid for previous versions, but I keep getting WolfSSL compile errors, and bring in a dependency 2. Select the diamond LIB_CRYPTO from the presentation layer from this document.

Microchip Video from their YouTube Channel. MPLAB® Harmony Configurator Tutorial - Creating a TCP/IP Project from Scratch
https://youtu.be/EyclfQkgLnM

NiyasMCHP, 2022-01-13:
Please refer to the following link to create a Harmony 3 TCP/IP project from scratch / migrating from Harmony 2.
https://github.com/Microchip-MPLAB-Harmony/net/wiki

You could also try the demo projects available in the 'net_apps_pic32mz' repository.

closed 2022-02-21 by adrian-aur

mhoover-fdi, 2022-02-21:
I opened a microchip ticket and it didn't work for them either.