Microchip-MPLAB-Harmony/net #33: SMI (management) for LANxxxx

gpmanly, 2022-02-07:
Is there a plan to support SMI (management) for LANxxxx devices? This would be very helpful in configuring 3-port switches like LAN9303, and LAN9353?

adrian-aur, 2022-02-21:
The MIIM driver could be used  to talk to the PHY for read/write of the control/status registers.
The MAC driver uses it to configure the PHY, etc.
What exactly is missing?

gpmanly, 2022-04-28:
Thank you for the reply, Adrian.
If I remember correctly, MIIM has limited access to the registers of LAN93xx devices while SMI can access all the registers. There is a workaround though, use the MIIM driver to emulate an SMI protocol.

adrian-aur, 2022-04-28:
There is a difference between MIIM and SMI at the hardware level and I don't think it's possible to use MIIM for SMI.
However, you normally get access to all registers if you use I2C for example.
So, a MIIM type driver over I2C is relatively easy to do - the MIIM driver is a virtual one, just need to replace internal operations with I2C ones.

gpmanly, 2022-11-08:
Found this article. I think this is the solution.
https://microchipsupport.force.com/s/article/Using-Harmony-3-MIIM-Driver-for-SMI

closed 2022-11-08 by gpmanly