Microchip-MPLAB-Harmony/net #34: mpfs_net.jar Version MPFS 3.6.0

Cargr, 2022-02-19:
The typedef struct HTTP_APP_Initialize is wrong declared in http_net_print.h.

Generated file:
```
//An entry for the dynamic variables which is used during the dynamic variable processing
typedef struct
{
    uint16_t    busy;           // buffer is currently in process
    uint16_t    bufferSize;     // size of the associated buffer
    char        data[HTTP_APP_DYNVAR_BUFFER_SIZE];  // buffer for writing the dynamic variable callback content
}HTTP_APP_DYNVAR_BUFFER;
```

Should be:
```
typedef struct HTTP_APP_DYNVAR_BUFFER
{
    uint16_t    busy;           // buffer is currently in process
    uint16_t    bufferSize;     // size of the associated buffer
    char        data[HTTP_APP_DYNVAR_BUFFER_SIZE];  // buffer for writing the dynamic variable callback content
}HTTP_APP_DYNVAR_BUFFER;
```

Using it from **Version 3.7.4**.

adrian-aur, 2022-02-21:
You probably refer to the HTTP_APP_DYNVAR_BUFFER data structure. What is the error, please explain?

Cargr, 2022-02-22:
Hello,

yes i refer to the HTTP_APP_DYNVAR_BUFFER, i use following function from Microchip Demo (custom_http_net_app.c).

```
void TCPIP_HTTP_NET_DynAcknowledge(TCPIP_HTTP_NET_CONN_HANDLE connHandle, const void *buffer, const struct _tag_TCPIP_HTTP_NET_USER_CALLBACK *pCBack)
{
    HTTP_APP_DYNVAR_BUFFER *pDynBuffer = (HTTP_APP_DYNVAR_BUFFER*)((const uint8_t *)buffer - offsetof(struct HTTP_APP_DYNVAR_BUFFER, data));

    pDynBuffer->busy = 0;
}
```

After generating a images with mpfs_net.jar i get following error:

`error: invalid use of undefined type 'struct HTTP_APP_DYNVAR_BUFFER'`

I did some test and also tried to regenerate the image from the demo pages, the error was the same, after comparing the generated and Harmony file i saw the missing tag.

Another solution would be to change the function from:

```
void TCPIP_HTTP_NET_DynAcknowledge(TCPIP_HTTP_NET_CONN_HANDLE connHandle, const void *buffer, const struct _tag_TCPIP_HTTP_NET_USER_CALLBACK *pCBack)
{
    HTTP_APP_DYNVAR_BUFFER *pDynBuffer = (HTTP_APP_DYNVAR_BUFFER*)((const uint8_t *)buffer - offsetof(struct HTTP_APP_DYNVAR_BUFFER, data));

    pDynBuffer->busy = 0;
} 
```

to:

```
void TCPIP_HTTP_NET_DynAcknowledge(TCPIP_HTTP_NET_CONN_HANDLE connHandle, const void *buffer, const struct _tag_TCPIP_HTTP_NET_USER_CALLBACK *pCBack)
{
    HTTP_APP_DYNVAR_BUFFER *pDynBuffer = (HTTP_APP_DYNVAR_BUFFER*)((const uint8_t *)buffer - offsetof(HTTP_APP_DYNVAR_BUFFER, data));

    pDynBuffer->busy = 0;
}
```

Removing **struct** keyword in **offsetof()**.

Hope this could help.

adrian-aur, 2022-02-22:
This has been fixed some time ago.
Please update your code with the latest revision.

closed 2022-02-22 by adrian-aur

Sergiadis, 2022-03-03:
Hello @adrian-aur 
We are having the same issue and we have downloaded the latest version of MPFS_NET from the repo.
What do you mean by:

> Please update your code with the latest revision.

Which code are you referring to? If this has been fixed, why isn't it on the repo?