Microchip-MPLAB-Harmony/net #37: Thread Safety

marcFriesen, 2022-04-11:
I am having an issue where the TCPIP task is overwriting other tasks stacks. The function `static uint16_t _TCPSocketTxFreeSize(TCB_STUB* pSkt)` is returning a size of 0xFFFF due to the function getting preempted after the second if statement, but before it returns the calculated size. At the time of comparing the head to the tail, the head is less, after the if is skipped the head and tail are equal. Here is the function body:

```
static uint16_t _TCPSocketTxFreeSize(TCB_STUB* pSkt)
{
    // Unconnected sockets shouldn't be transmitting anything.
    if(!( (pSkt->smState == TCPIP_TCP_STATE_ESTABLISHED) || (pSkt->smState == TCPIP_TCP_STATE_CLOSE_WAIT) ))
    {
        return 0;
    }

	
    // Calculate the free space in this socket's TX FIFO	
    if(pSkt->txHead >= pSkt->txTail)
    {
        return (pSkt->txEnd - pSkt->txStart - 1) - (pSkt->txHead - pSkt->txTail);
    }

    return pSkt->txTail - pSkt->txHead - 1;
}
```

Is thought the library was thread-safe, is there something else I need to add? I've configured the library to use thread-safe malloc/free provided by FreeRTOS, as well as adding a thread-safe calloc.

marcFriesen, 2022-04-11:
No, the socket is opened and used by one thread only. I have a while loop that is writing a buffer (~4K) until all bytes have been sent. The TCPIP task preempts this task and seems to change the head/tail pointers while we are also trying to send more data. I've added a workaround to delay 5ms in the while loop between successive reads/writes of TCP which seems to avoid this race condition.

adrian-aur, 2022-04-11:
It'd be helpful if you could identify the part of the code that overwrites those head/tail pointers.
Are you sure it's the TCP/IP stack? Maybe you could increase the stack size for some of the threads in there.
I think it's important to understand if this is a buffer overwrite or simply a 'legitimate' access made by the stack, but without the necessary multi-thread protection.

Also, would it be possible to send to me a simplified version of your code that exposes this behavior?
I'll try to dig into it and see what happens.
But it may take a while, it could be not obvious how to catch this kind of behavior.