Microchip-MPLAB-Harmony/net #41: WolfMqtt reconnect attempts with wrong Gateway Settings

Elpra-Bozas, 2023-02-13:
Dear support,
we are facing an issue while trying to handle a reconnect attempt to an mqtt broker and ARP is not resolved due to wrong net settings (e.x. wrong Gateway). With a wrong Gate way we get a lot of "IPv4: ARP entries pool empty!" console messages and in our state machine we are facing a lot of Heap allocation problems. (I can throw specific state machine code if necessary).
Is there a workaround? Is it possible to retrieve default gateway IP before enabling any TCP service where Gateway settings are mandatory?

adrian-aur, 2023-02-13:
Please help me understand: why do you have a wrong gateway setting?
Without a proper gateway address you won't be able to communicate anyway - except in your LAN.
 
About the heap problems, please post the results of the 'heapinfo' command so we can take a look.

Elpra-Bozas, 2023-02-13:
We have a project where boards are installed on different premises each having their own net settings, so net configuration is not standard. The configuration is not performed by us, so the device in our case might easily be left with a Gateway setting that is not correct. We have a web-server that can run in the corresponding LAN and can update - retrieve part of the application specific user logic data from that interface, but we also have an Mqtt connection where we run the entire application status - configuration from the Mqtt interface.  We also have an option to connect locally (in our LAN as you mentioned ) to a host again via Mqtt, by opening an Mqtt server on the host and triggering the board to connect to that server (let's call this connection scheme "Local Mqtt"). Our main problem at the moment is the scenario where someone has left the board with a wrong Gateway (and where clearly we can't connect to a remote Mqtt Server) and we want to trigger our Mqtt connection locally through the LAN. Leaving the board to remote Mqtt connection requests forces the ARP entries pool empty message after a while and we cannot seem to find the proper way to handle the swich from the remote Mqtt connect request to the local Mqtt connect request.

adrian-aur, 2023-02-13:
There are few things that you can do:
- increase the IPv4 initialization data TCPIP_IPV4_MODULE_CONFIG::arpEntries
         - Note that it should be <= the number of total ARP cache entries for all interfaces so adjust the TCPIP_ARP_MODULE_CONFIG::cacheEntries
         - Anyway, check that the number of arpEntries corresponds with cacheEntries. Probably you have a lots of hosts that you have to solve in your app.
- Decrease the ARP timeout: TCPIP_ARP_MODULE_CONFIG::entrySolvedTmo and entryPendingTmo.
        - This way the entries will be removed quicker, including the ones that cannot be solved
- Increase the heap memory. How much do you have right now? Post the result of a 'heapinfo' command
-

adrian-aur, 2023-02-13:
Another thing: do you check for TCPIP_STACK_NetIsReady() before starting transactions?
And to answer your previous question: you can delay the start of any (TCP) connection until you have a status that is stable. There should be API for all the settings.

Elpra-Bozas, 2023-02-14:
We are using PIC32MZ1024 series for this project, We have set the System Heap Size to 120K and TCPIP CORE Heap to internal for 70K (TCP Heap Size estimation on Harmony is 67K). heapInfo cmd throws the following:
heapinfo
Heap type: internal. Initial created heap size: 69904 Bytes
Allocable block heap size: 7520 Bytes
All available heap size: 18000 Bytes, high watermark: 53648
Last heap error: 0x0
Trace info: 
Module:    1, totAllocated: 264848, currAllocated: 18672, totFailed:     0, maxFailed:     0 
Module: 4160, totAllocated:  1728, currAllocated:   864, totFailed:     0, maxFailed:     0 
Module:    2, totAllocated:   256, currAllocated:   256, totFailed:     0, maxFailed:     0 
Module:    3, totAllocated:   176, currAllocated:   176, totFailed:     0, maxFailed:     0 
Module:    9, totAllocated:   624, currAllocated:   624, totFailed:     0, maxFailed:     0 
Module:   10, totAllocated: 972208, currAllocated: 25184, totFailed:     0, maxFailed:     0 
Module:   12, totAllocated:   112, currAllocated:   112, totFailed:     0, maxFailed:     0 
Module:   15, totAllocated:   944, currAllocated:   944, totFailed:     0, maxFailed:     0 
Module:   24, totAllocated:  5072, currAllocated:  5072, totFailed:     0, maxFailed:     0 
Module:   14, totAllocated:  1056, currAllocated:     0, totFailed:     0, maxFailed:     0 

At the moment we use a small persist of 3 times with a delay of 5 sec between them to ping the Gateway and if it fails then we stop the remote Mqtt connection and waiting for a local mqtt connect event. We are noticing that even if we wait for a long time in this IDLE state then the console is streamed like crazy (possibly trying to print a heap allocation heap error on some module but the entire string does not manage to get printed correctly, e.x. "allHeap allHeap allHeap allHeap allHeap allHeap allHeap allHeap allHeap allHeap"). **Is there any issue with the ARP module trying to handle the empty chase entries after a significant amount of time?** (there is no way i can further do anything else with the console at this point, heapinfo or any other debug info is not accessible)

Note: we have tried to increase the ARP max entries with no luck, we just managed to extend the time before the problem arises. 
Yes we are using the TCPIP_STACK_NetIsReady before moving on with all user code that handles TCP services.
Seems we are running out ot RAM so increase of heap is not  a straight forward action at the moment.
I'll try to decrease the ARP timeout and will come back.

Elpra-Bozas, 2023-02-15:
Hi Andrian, we went through the modifications with the same behavior. I forgot to mention that we are using Harmony 3 and the net library version we are using is v3.8.0. We see that there is an update on this lib, but we haven't made the transition yet since this is supported only on MCC based projects and the migration from ahrmony 3 to MCC does not work.
Is there a chance there are updates on ARP or something that sounds related to our issue with the last .net library?

adrian-aur, 2023-02-15:
No need to migrate to v3.9.0, no relevant changes for your problem.
First you should try to fix the console problem, so you can get the messages printed out correctly:
- increase the console/command processor buffer
- increase the UART TX buffer

Eliminate all those "allHeap..." messages that flood the console. Identify where they come from and suppress them.
Let the board run for a while, until the condition is reached.
 - Then try the 'arp eth0 list' and let's see the output
 - Same for 'heapinfo'
Hopefully we'll find out more of what's going on.

closed 2024-09-05 by adrian-aur