Microchip-MPLAB-Harmony/net #42: GMAC issue PIC32CX

TheHogfather562, 2023-05-03:
Hi there,

When adding the GMAC driver into the project, nothing shows up in the project graph. Also the TCP/IP Configuration Window won't open.

Are the PIC32CX not supported yet? I can't see any reference to them in the driver src files https://github.com/Microchip-MPLAB-Harmony/net/tree/master/driver/gmac/src/dynamic

I have attached a simple project to demonstrate the problem.

[GMAC_Test.zip](https://github.com/Microchip-MPLAB-Harmony/net/files/11381322/GMAC_Test.zip)

If it's not yet supported, when will support be available?

Kind regards

adrian-aur, 2023-05-08:
PIC32CX support is not yet available. We're working on it and we'll release in a few weeks.

TheHogfather562, 2023-05-09:
Thank you for letting me know! I look forward to the release!

TheHogfather562, 2023-05-24:
Any news on when this support will be released?

adrian-aur, 2023-05-24:
The MCC support has been added.
Demo app is under test now.
Probably 2-3 weeks to release.

TheHogfather562, 2023-06-20:
Any update on this?

adrian-aur, 2023-06-20:
The release is under way, we're performing the tests. Either the end of this week or the next one.

TheHogfather562, 2023-07-10:
Has this been released yet?

Also when you said this:

> The MCC support has been added.
> Demo app is under test now.
> Probably 2-3 weeks to release.

Did you mean that it should work now and that you're just developing demo apps for it. Or that it won't work my end still until the demo app is released?

adrian-aur, 2023-07-10:
Hasn't been released yet. Sorry, sometimes it takes longer than estimated. We'll try to do it as soon as possible.
There will be also demos for PIC32CX.

closed 2023-07-31 by TheHogfather562