Microchip-MPLAB-Harmony/net #44: my application crashes

picerno76, 2023-08-03:
I have an application on a WFI3201 that sends UDP packets. I'm using the net package v.3.8 and my application crashes after about 20 seconds when there is no packet sending or if destination address doesn't exist. I have found that it happens because at line 2428 in the file ipv4.c, pHeader is NULL so, if I add a break in this condition, it doesn't crash anymore

adrian-aur, 2023-08-21:
Sorry for the delay, somehow I've missed your post.
The line #2428 sets the pHeader = pRxPkt->pNetLayer;
On a RX packet, the pNetLayer needs to be set by the MAC driver.
Apparently there's an issue in the Wi-Fi driver.
Please give some more details about the Wi-Fi distribution and application that you use so that I can forward this issue to the corresponding team. A way to reproduce the issue would be nice.

adrian-aur, 2023-08-22:
OK, I've forwarded your issue to the Wi-Fi team.

himanshuseth-microchip, 2023-08-28:
1. The crash is coming in the RX path in the TCP IP Stack
2. We have run both are UDP Client demo and the UDP Server demo and in both cases, we open a UDP socket, but in neither of the cases we crash even after we have waited for 2 minutes also. 
3. We have sent a packet from a PEER UDP client to the UDP server running on Rio2 with both incorrect and correct ports to see if we experience a crash in RX Flow which is being observed by you, We do not see any. 
4. In the above point, on receiving a packet on the correct port, we even echo back the same packet to the peer UDP client and there is NO crash seen here too.
5. Was any existing Rio2 demo used for creating this application or the application was built from scratch ?
6. Can the you please share your application code for us to check ?

picerno76, 2023-08-28:
Hi,
I have tried again and how I thought, I have found that it only happens with Mikrotik AP and not with a D-Link AP. This is probably why you don't have the issue. I have seen that pNetLayer is assigned in _TCPIP_PKT_PacketAlloc if segLoadLen > 0. In my case segLoadLen is 0 because in DRV_PIC32MZW_MACEthernetSendPacket lengthEthMsg is 14, so lengthEthMsg-ETHERNET_HDR_LEN == 0. By then, I don't see where DRV_PIC32MZW_MACEthernetSendPacket has been called.
Thank you

himanshuseth-microchip, 2023-08-29:
Hi,

Could you please raise a Salesforce ticket for this.

Thanks.

himanshuseth-microchip, 2023-09-04:
Hi,
Sorry for the delayed response.
You can create a login and raise the ticket on https://www.microchip.com/wwwregister/login.aspx?ReturnUrl=https://microchipsupport.force.com
There is no fees.

adrian-aur, 2023-09-18:
Well, even if it doesn't happen anymore with the keep-alive disabled, still there is something wrong with these packets, and the crash should not occur.
Probably of your zero-length packets?
Do you use the TCP/IP stack keep-alive mechanism or you do your own stuff?

picerno76, 2023-09-18:
it doesn't concern the TCP/IP keep-alive mechanism because it should be at a lower layer, data link or physical

closed 2024-09-05 by adrian-aur