Microchip-MPLAB-Harmony/net #47: Can't migrate from 3.12.0 to 3.12.1

krzysztofstrycharz, 2024-08-07:
I'cant migrate MCC project from library version 3.12.0 to 3.12.1 - SAMA7G54 target. Everythink working fine up to commit d6575be.

mohammedfahadr-mchp, 2024-08-19:
Hi @krzysztofstrycharz,
Can you please share the issue? what issue do you see
Also, please share the steps to reproduce the issue.

hitac10, 2024-08-22:
The same symptom occurs with the PIC32MZ target.
Updating net v3.11.1 to v3.12.0 works fine, but updating to v3.12.1 or v3.12.2 causes [Project Grapth] to display blank.

This is reproduced with the [StarterKit_EF_rev2] project attached to this forum.
https://forum.microchip.com/s/topic/a5CV40000000Td7MAE/t394295?comment=P-2963903

immfr, 2024-08-23:
Hi,
Try doing this, it works for me.

First check if you've installed & using the latest MCC plugin 5.5.1
Next, open the .mc3 file which is present inside the project folder (.X folder)
- find the keyword **coreVersion** and ensure it is "5.7.1"
- find the keyword **ILibraryFile** and look at the end of the line, change the version to "1.5.2"

(basically, updating these versions to the latest)

Save the .mc3 file.
Start/open the MCC now. 
Project graph opens correctly.

Maybe it will work for you, try once.

krzysztofstrycharz, 2024-08-23:
The issue was described by @hitac10 . After updating net library to 3.12.1 or 3.12.2 version the MCC project, which was created a year ago, opens without any significal error but project graph is blank and .mc3 file is corrupt. 

Blank project graph.
[image]

Corrupt .mc3 file.
[image]

After change ILibraryFile version to 1.5.2 the MCC opens project successfully, thank you @immfr.
`<ILibraryFile class="com.microchip.mcc.core.library.BaseLibraryFile" libraryClass="com.microchip.mcc.harmony.Harmony3Library" version="1.3.1"/>`

hitac10, 2024-08-23:
Hi @immfr,

> * find the keyword **ILibraryFile** and look at the end of the line, change the version to "1.5.2"

Thank you for the useful information!!
I also changed the ILibraryFile version in the .mc3 file to 1.5.2, and MCC was able to open the project successfully.

It seems to be a problem with MCC, so some kind of solution is needed.

hitac10, 2024-08-23:
I thought manually editing the .mc3 file was an unofficial method, but I found it in the official Knowledge.

[Blocks/Components are NOT shown in the Project Graph of MCC Harmony V3 project
](https://microchipsupport.force.com/s/article/Blocks-Components-are-NOT-shown-in-the-Project-Graph-of-MCC-Harmony-V3-project)

immfr, 2024-08-26:
I'm glad that it worked for you :)

closed 2024-09-03 by adrian-aur