Microchip-MPLAB-Harmony/net #48: MCC / Harmony default setting for TCPIP_STACK_INTERFACE_CHANGE_SIGNALING is false

nwright-mcc, 2024-08-15:
The Manager Module help at https://microchip-mplab-harmony.github.io/net/GUID-B37C4F4C-DC2D-48D9-9909-AACBA987B57A.html states that the default setting for this setting is true.  However, MCC v5.5.1 initializes this to false when using the TCP/IP Configuration plugin.  

This resulted in incorrect behavior on a network where the DHCP client timed out and used the default IP address and the TCP socket was opened with this address, but the DHCP server subsequently assigned an address to the device.  The TCP socket did not update to the new address.

adrian-aur, 2024-08-15:
Can you please describe what exactly setting is the problem. It may be just some documentation error or the setting has been changed from its default value. But to check it I need to know which one you're referring to.

nwright-mcc, 2024-08-15:
In MCC, it is set under TCPIP CORE - Advanced Settings - Enable Interface Address Change Signaling.

adrian-aur, 2024-08-15:
Yes, it's a documentation issue. The default value is false.
I'll make the request to change it.
However, it's always a good idea to verify the settings of your project, especially the 'advanced' ones, if your app is relying on them. Especially when starting from another app and then add functionality to it.

Please note that if this feature is enabled the client sockets will be killed and the server sockets will be disconnected.
For client sockets the app intervention is necessary anyway.

adrian-aur, 2024-09-05:
An internal request has been made to update the documentation for this.

closed 2024-09-05 by adrian-aur